DIYOSEPP/EspUnoWiFi #1: Arduino Uno not being reset with Feather Huzzah ESP8266

kabrio, 2019-06-24:
Hi,

I installed the software on ESP8266 Huzzah board and can connect to it.
When uploading to Arduino Uno using Arduino IDE (selecting the network port) the upload fails 
as the Arduino is not being reset. 
```
Resetting the board
Failed to reset the board, upload failed
```

What am I doing wrong?

Kashyapkoshti, 2019-12-13:
i am getting upload successfully but in flashing gets error in arduino uno.

DIYOSEPP, 2019-12-13:
Hello, 

We don’t sell Arduino WiFi or modules to make it WiFi enabled. Unfortunately I do not have experience with this and am unable to offer assistance. 

Daniel | OSEPP Senior Designer & Sales
OSEPP Electronics inc | http://osepp.com/ 
mobile: (786)-253-4040
Email: danny@osepp.com

kabrio, 2020-02-13:
Thank you for the reply!

Wilhelm1Daniel, 2020-03-25:
Be Sure to select the correct (reset) pin in your board. The Code is written to gpio 2.