DIYOSEPP/EspUnoWiFi #2: OTA programming in arduino mega

Kashyapkoshti, 2019-12-13:
i want to do this with arduino mega. can you help me with that?

Wilhelm1Daniel, 2020-03-26:
I Think it's not really a problem.
Connect the Mega like an Uno.

I do the same with an Arduino Pro Mini...
The only thing I have to change is the baudrate in the ESPUNOWIFI.INO for the Pro Mini to 57600

> int entryBootloader()
{
  Serial.begin(**57600**);
  pinMode(resetPin, OUTPUT_OPEN_DRAIN);
  digitalWrite(resetPin, LOW);`

Kashyapkoshti, 2020-03-27:
So for arduino mega which baud rate is preferable?

Wilhelm1Daniel, 2020-03-27:
I'm not sure. Try to program the MEGA via USB to TTL-Adapter on Serial1 ( pin 18 and 19).  In the log of Arduino IDE you Should see if the baudrate changed to Optimum speed.-- Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

Kashyapkoshti, 2020-08-28:
@Wilhelm1Daniel did you try this for arduino mega 2560?

youssiefkhaled, 2023-09-08:
> @Wilhelm1Daniel did you try this for arduino mega 2560?

did it work for you?