DIYOSEPP/EspUnoWiFi #11: Board connected to wifi but does not get IP address

brunoaduarte, 2023-07-11:
Hi @DIYOSEPP I'm trying to use the EspUnoWiFi with my Osepp Uno R3 Plus (ATmega328P) + ESP01 (ESP8266) shield.
https://osepp.com/electronic-modules/microcontroller-boards/105-osepp-uno-r3-plus

I can successfully upload the sketch to the ESP8266, but the board does not get any IP from my router.

```
Local IP	(IP unset)
subnetMask	(IP unset)
gateway	        (IP unset)
dns	        (IP unset)
```

fsinnes, 2023-08-07:
Did you figure this out? I am also getting an incorrect upload line

brunoaduarte, 2023-08-09:
> Did you figure this out? I am also getting an incorrect upload line

Just use an older IDE of Arduino, like 1.8.X. Seems there's a bug with the parameters parsing on the newer versions.